grupacosmo/cansat #37: Setup accelerometer

wiktorwieclaw, 2023-02-25:


wiktorwieclaw, 2023-02-26:
See `embedded-hal-bus`, it allows sharing I2C in a safe manner.

wiktorwieclaw, 2023-02-26:
[embedded-hal-bus](https://github.com/rust-embedded/embedded-hal/tree/master/embedded-hal-bus) doesn't support bus hsaring yet. For the time being, we can use [setup-bus](https://github.com/Rahix/shared-bus). See [shared-bus-example](https://github.com/ryan-summers/shared-bus-example).

wiktorwieclaw, 2023-02-26:
Since I already got a hang of `shared_bus` I actually went ahead and made a [PR](https://github.com/grupacosmo/cansat/pull/41) that inits the shared bus. Feel free to continue working on it.

Bartosz-Slowik, 2023-03-16:
I used your suggestion but wrote it based on newer main version.

closed 2023-03-23 by wiktorwieclaw