grupacosmo/cansat #50: Print PCB

wiktorwieclaw, 2023-04-02:
Issue for tracking PCB progress

Bartosz-Slowik, 2023-04-03:
I made an open project with easyEDA so anyone can make suggestions.
https://oshwlab.com/cansat-rust/cansat
[image]

Bartosz-Slowik, 2023-04-03:
Aktualny schemat
[image]

Bartosz-Slowik, 2023-04-03:
By Kamil suggestion Lora will be SDM mounted so I added a spare uart 4 pin for development without soldering

Bartosz-Slowik, 2023-04-04:
I got permission to order a part from local manufacturer. 
I need to fix his concerns + add buzzer + (maybe rtclock)
Avrage order should take 3 days but there are easter holidays comming.

Bartosz-Slowik, 2023-04-27:
First PCB arrived and all connections are working perfectly. At least we have a fallback to launch the cansat if newer pcb has a problem.

wiktorwieclaw, 2023-05-14:
@Bartosz-Slowik we can close this one, right?

Bartosz-Slowik, 2023-08-05:
We made One more revision for PCB:
We will no longer work with local manufacturers:
[image]