grupacosmo/cansat #51: Design cansat housing

wiktorwieclaw, 2023-04-02:
Tracking issue for the process of cansat housing design

Concerns:
- [] How to deal with antennas?
- [] How to attach the parachute?

Feel free to add more things to the list.

Bartosz-Slowik, 2023-04-04:
Cansat max dimensions are 110mm tall with 66mm diameter.
3d printed housing with 4 mm wall will be sufficient.
We need a groove on the side for 1/4 dipol antenna for LORA
We need a top "window" for GPS ceramic antenna

We need 8 loops/hooks for mounting parachute strings.

We need to fit pcb + separete gps +  18650 cell box inside

We have 3 ideas of mounting pcb
  -1) empty can with pcb wprapped in bubble wrap
  -2) printing two halves horizontally with a platform to screw pcb into
  -3) printing open can with a lid that has a long platform for pcb

Bartosz-Slowik, 2023-04-27:
I assigned a new member to help with this issue. 
I designed the housing + lid approach, but have a problem with 3d printer.

Bartosz-Slowik, 2023-05-25:
Done

closed 2023-05-25 by Bartosz-Slowik