grupacosmo/cansat #53: Connecting to STM32 without ST-LINK

Bartosz-Slowik, 2023-04-06:
We have 2 STM boards that have broken ports for our standard connection.

I saw there are at least 2 alternative ways of flashing this STM but i dont think debugging would work. 
Maybe someone can test or research this more.

wiktorwieclaw, 2023-05-13:
Run:
```
cargo install cargo-dfu
```
Connect your board to USB, hold BOOT0 button, press NRST button and release BOOT0 button. This should run the MCU in bootloader mode.

Change directory to the project you want to flash, e.g
```
cd crates/cansat-stm32f4
```
and run 
```
cargo dfu
```

### Troubleshooting
If your device is not properly recognized by your system, try to disconnect all serial peripherals and repeat.

wiktorwieclaw, 2023-05-13:
Let's leave this issue open, it's probably a good idea to update the cansat book #60 with this information.