grupacosmo/cansat #58: Gateway crate

wiktorwieclaw, 2023-04-14:
Tracking issue for Gateway development.

let's create a new binary crate in the `tools/` directory

We will have to communicate with lora via USB using [AT commands](https://files.seeedstudio.com/products/317990687/res/LoRa-E5%20AT%20Command%20Specification_V1.0%20.pdf).

Currently cansat sends measurements in CSV format, but we will migrate to a binary format from [postcard](https://crates.io/crates/postcard) crate. So we will need to use postcard to deserialize the data into Rust types

Recommended crates to use:
* [clap](https://crates.io/crates/clap) - command line argument parser, we use it in in [xtask](https://github.com/grupacosmo/cansat/tree/main/xtask). Please use it's derive macros.
* [color-eyre](https://crates.io/crates/color-eyre) - error chaining and reporting. any error can be converted to `eyre::Report` and you can add additional context to errors with `wrap_err`
  ```rust
  fn f() -> Result<(), eyre::Report>
      may_fail().wrap_err("Failed to do a thing")?;
      may_also_fail().wrap_err("Failed to do another thing")?;
      Ok(())
  }
  ```
* [rusb](https://crates.io/crates/rusb) - a thin wrapper around libusb, use it to communicate with lora
* [lora-e5](https://crates.io/crates/lora-e5) - lora-e5 support library based on [tokio](https://crates.io/crates/tokio) async runtime. It's hasn't been used very much, so I cannot guarantee it will work well. If not then we will have to handle lora ourselves.
*  [postcard](https://crates.io/crates/postcard) - deserialize data sent from cansat

Let's try to write down all of the upcoming questions and decisions in this thread.

wiktorwieclaw, 2023-04-14:
I'd suggest not to focus on creating clean LoRa abstractions at first, let's implement an ugly prototype that works. We are short on time, we will clean it up later.

wiktorwieclaw, 2023-04-14:
my initial suggestion is to have one thread read bytes from the serialport into a buffer in a loop. When it reaches the '\r\n` it consumes the buffer and sends the message to another thread though a [channel](https://doc.rust-lang.org/std/sync/mpsc/index.html). The other thread will parse the messages and print them to the screen

wiktorwieclaw, 2023-04-14:
Message format is specified here: https://github.com/grupacosmo/cansat/blob/a20da23e0b0c147d7845646307a128eef371ee45/crates/cansat-core/src/measurements.rs#L6-L24
Just copy the struct definition, change the field types to primitives (eg. `f32` instead of `Temperature`) and derive `serde::Deserialize`

wiktorwieclaw, 2023-04-21:
example logs in csv
```
24.209528,97702.17,304.03323,"$GPGLL,,,,,201513.00,V,N*4E",16,272,3920,face down
24.209528,97702.0,304.04645,"$GPGLL,,,,,201513.00,V,N*4E",32,288,3936,face down
24.209528,97702.0,304.04645,"$GPGLL,,,,,201513.00,V,N*4E",64,272,3952,face down
24.209528,97702.0,304.04645,"$GPGSA,A,1,,,,,,,,,,,,,99.99,99.99,99.99*30",48,224,3968,face down
24.209528,97702.0,304.04645,"$GPGSV,3,3,12,21,37,154,,28,30,063,,31,33,089,,32,06,048,06*7F",32,272,3952,face down
24.209528,97702.0,304.04645,"$GPGLL,,,,,201514.00,V,N*49",64,272,3872,face down
24.209528,97702.0,304.04645,"$GPGLL,,,,,201514.00,V,N*49",64,224,3936,face down
24.212046,97702.4,304.01474,"$GPGLL,,,,,201514.00,V,N*49",112,288,3968,face down
```

vanderAries, 2023-04-30:
Added new branch: [pbak/py-lora](https://github.com/grupacosmo/cansat/tree/pbak/py-lora) with some transmitter and receiver programs

wiktorwieclaw, 2023-05-12:
I merged the stuff from pbak/py-lora into main. We still have to rewrite it in Rust, so I'll keep the issue open.

vanderAries, 2023-05-22:
Added new functions:

-  connection test: function sends "AT" command, to check if device is connected and working. One can run this using `cargo lora-cli conntest --port {port}`. @wiktorwieclaw not sure if conntest name is ok for this
- function for sending custom commands (stored in `cansat/tools/lora-cli/commands.txt`) in case of any adjustments like frequency change etc.

vanderAries, 2023-05-22:
TODO: 

1. Update README with new functionality
2. Add parsing function for displaying received information in readable form, together with SNR and RSSI values

closed 2023-08-28 by Bartosz-Slowik