grupacosmo/cansat #68: CanSat 2.0 Wishlist

wiktorwieclaw, 2023-05-18:
- [ ] try to migrate from rtic to embassy
- [ ] peripheral health monitoring system
- [ ] peripheral recovery system, that can power off peripherals to reset them
- [ ] movement and position recognition system
- [ ] mission control system - track whether the flight has already started, whether the CanSat is falling and when it has landed, then we can dispatch different tasks depending on it's state

I'm open to proposals in the comments

vanderAries, 2023-05-23:
Ground Station for two way communication with Cansat. It could be used either with Cansat or with rocket flight computer.

wiktorwieclaw, 2023-06-07:
I decided not to move to embassy after all. Their stm32 HAL didn't have a release yet and on the latest revision it depends on many crates that we will not need such as `embassy-net` which is bonkers.

RTIC had a 2.0 release recently and that alone will give us a huge win for ergonomics with minimal effort.